ternaris/marv-robotics #3: Feature request: Workflow for removing filesets

NikolausDemmel, 2016-10-10:
I am in the situation that I added some bags that I now want to ignore with `.marvignore`. So for it is only possible to remove them if you know the uuid. It would be great if the cli would support some workflow for finding the current filesets. 

Ideas:

```
$ marv fileset list
./scanroot/foo-bar <uuid>
./scanroot/quux-quax <uuid>
...
```

```
$ marv  fileset discard `marv fileset list scanroot/subfolder/ --uuid`
```

```
$ marv fileset discard --ignored
```

NikolausDemmel, 2016-11-04:
This is related also to #9, for which one also needs a way to select a subset of bagfiles for which to rerun a set of changed nodes.

chaoflow, 2018-01-31:
@NikolausDemmel to remove datasets in specific directories:

```
marv query --path path/to/ignored/directory |xargs marv discard
```

Regarding your Update, I'll create a new issue.

closed 2018-01-31 by chaoflow