ternaris/marv-robotics #6: beta install on 14.04 with indigo

NikolausDemmel, 2016-10-10:
When following the install instructions on my 14.04 machine with ROS indigo, I get the following error after marv was installed:

```
$ marv
Traceback (most recent call last):
  File "/home/den2pal/work/bagbunker/marv-venv/bin/marv", line 5, in <module>
    from pkg_resources import load_entry_point
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2749, in <module>
    working_set = WorkingSet._build_master()
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 446, in _build_master
    return cls._build_from_requirements(__requires__)
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 459, in _build_from_requirements
    dists = ws.resolve(reqs, Environment())
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 628, in resolve
    raise DistributionNotFound(req)
pkg_resources.DistributionNotFound: python-dateutil>2.2
```

I solved this by running the following from within the venv:

```
pip install -U python-dateutil
```

NikolausDemmel, 2016-10-10:
I also have the following additional issues with this setup (in a Docker image):

```
$ marv node run
Traceback (most recent call last):
  File "/marv/venv/bin/marv", line 5, in <module>
    from pkg_resources import load_entry_point
  File "/marv/venv/local/lib/python2.7/site-packages/pkg_resources.py", line 2720, in <module>
    parse_requirements(__requires__), Environment()
  File "/marv/venv/local/lib/python2.7/site-packages/pkg_resources.py", line 592, in resolve
    raise VersionConflict(dist,req) # XXX put more info here
pkg_resources.VersionConflict: (setuptools 3.3 (/usr/lib/python2.7/dist-packages), Requirement.parse('setuptools>=18.5'))
```

--> `pip install -U setuptools`

And for the system dependencies I needed to add `apt-get install libffi-dev`, else I would get compiler errors during `pip install marv-robotics`.

chaoflow, 2016-10-20:
Thanks a lot for the feedback! There is a PPA for latest Python 2.7.12 https://launchpad.net/~fkrull/+archive/ubuntu/deadsnakes-python2.7

Could you retry with that?

NikolausDemmel, 2016-10-30:
Thanks. 

Adding that PPA gets rid of python/pip errors and the need for manually upgrading `python-dateutil`.

On top of the install instructions from the readme, I still need to install `libffi-dev`, I need to now manually `pip install cffi`, else I get an error during `pip install marv-robotics`.

And ofc you still need to use `pip install --pre marv-robotics`.

My current setup steps (Dockerfile) can be found here: https://gist.github.com/NikolausDemmel/7281ae200f6c251b5695e96563ab60a9

chaoflow, 2016-10-31:
Thanks a lot! Instructions are updated: https://github.com/ternaris/marv-robotics/commit/79d912612eefb3e6f92b8beba15f0023dd3a0aa7

closed 2016-10-31 by chaoflow

NikolausDemmel, 2016-10-31:
Thanks! LGTM. I updated our Dockerfile: https://gist.github.com/NikolausDemmel/7281ae200f6c251b5695e96563ab60a9