ternaris/marv-robotics #11: Address already in use if running uwsgi --ini uwsgi.conf  after the host machine has crashed

dejanpan, 2016-12-05:
We had a case when marv host (ubuntu 14.04) ran out of disk space and had restart it.

After starting uwsgi.conf again we get "Address already in use"  error. Full stack trace below.
```
[uWSGI] getting INI configuration from uwsgi.conf
*** Starting uWSGI 2.0.14 (64bit) on [Mon Dec  5 02:11:43 2016] ***
compiled with version: 4.8.4 on 04 November 2016 23:25:39
os: Linux-3.13.0-100-generic #147-Ubuntu SMP Tue Oct 18 16:48:51 UTC 2016
nodename: ip-172-20-1-243
machine: x86_64
clock source: unix
pcre jit disabled
detected number of CPU cores: 8
current working directory: /home/ubuntu/site
*** running under screen session 4074.pts-0.ip-172-20-1-243 ***
detected binary path: /home/ubuntu/venv/bin/uwsgi
*** WARNING: you are running uWSGI without its master process manager ***
your processes number limit is 257330
your memory page size is 4096 bytes
detected max file descriptor number: 1024
lock engine: pthread robust mutexes
thunder lock: disabled (you can enable it with --thunder-lock)
uWSGI http bound on 0.0.0.0:5000 fd 4
spawned uWSGI http 1 (pid: 4288)
uwsgi socket 0 bound to TCP address 127.0.0.1:44949 (port auto-assigned) fd 3
Python version: 2.7.12 (default, Jul 18 2016, 15:07:05)  [GCC 4.8.4]
Python main interpreter initialized at 0xf4e9b0
python threads support enabled
your server socket listen backlog is limited to 100 connections
your mercy for graceful operations on workers is 60 seconds
mapped 144896 bytes (141 KB) for 8 cores
*** Operational MODE: threaded ***
2016-12-05 02:11:44,134 WARNING marv._site Unused config sections ['marv']
2016-12-05 02:11:44,137 DEBUG marv.app.listing init app
2016-12-05 02:11:44,195 INFO marv.app.listing listening on /home/ubuntu/site/notify
WSGI app 0 (mountpoint='') ready in 1 seconds on interpreter 0xf4e9b0 pid: 4287 (default app)
*** uWSGI is running in multiple interpreter mode ***
spawned uWSGI worker 1 (and the only) (pid: 4287, cores: 8)
2016-12-05 02:11:44,199 INFO marv.app.listing Listing is up-to-date
2016-12-05 02:11:44,200 INFO marv.app.listing waiting for notification
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/home/ubuntu/venv/local/lib/python2.7/site-packages/marv/app/listing.py", line 157, in listen
    listener = Listener(self.site.socket, authkey=self.site.authkey)
  File "/usr/lib/python2.7/multiprocessing/connection.py", line 132, in __init__
    self._listener = SocketListener(address, family, backlog)
  File "/usr/lib/python2.7/multiprocessing/connection.py", line 256, in __init__
    self._socket.bind(address)
  File "/usr/lib/python2.7/socket.py", line 228, in meth
    return getattr(self._sock,name)(*args)
error: [Errno 98] Address already in use
```

dejanpan, 2016-12-05:
Another artifact of this behavior is that the listing on the landing page does not get updated after 
` marv fileset scan`
and
`marv node run --all-filesets --all-nodes`
commands.

chaoflow, 2016-12-07:
It's the notification socket, that cannot be bound. This code is being refactored. As a workaround, stop uwsgi, remove the notify socket (in your case /home/ubuntu/site/notify), and start again. The listing will be updated and stay up-to-date afterwards.

chaoflow, 2017-11-17:
17.11 does not use a notification socket anymore. Instead cli and server access an sqlite database.

closed 2017-11-17 by chaoflow