ternaris/marv-robotics #12: Systemd unit

Sekuraz, 2016-12-22:
We are currently preparing to use marv in a development environment.

In order to increase maintainability of the service we would highly appreciate a systemd unit file for marv as well as (on the long term) a debian package.

NikolausDemmel, 2016-12-22:
We build docker image for marv and then run marv in container. Docker does a fine job of also making sure the container is restarted at system boot. I am happy to share our current dockerfile if that is of interest. You might tweak it to your needs though.

Sekuraz, 2016-12-23:
would be of interest, but we are currently using a virtualenv. Have you considered uploading your docker image to the docker hub?

gavanderhoorn, 2016-12-29:
@NikolausDemmel: have things significantly changed compared to your [gist](https://gist.github.com/NikolausDemmel/7281ae200f6c251b5695e96563ab60a9)?

NikolausDemmel, 2016-12-30:
@gavanderhoorn, no not substantially. Feel free to use that as a starting point (there is some stuff specific to our setup with proxy etc in there). What has changed is that I have since added support to add custom node packages and have them automatically updated when mounted during `docker run` (useful for development).

Also, we are now using the enterprise edition, so we are not installing from pypi but from source from a git submodules so there is some logic to install stuff in the correct order.

I should push all this to github. I hope I get round to it next week or so.

@Sekuraz I don't really plan to maintain a docker image for now. Maybe later if I get round to cleaning the dockerfile from all the custom stuff (proxy, certificates, ...). Note that I'm just a user and don't work for Ternaris. Maybe @chaoflow wants to maintain a base image at some point (feel free to use our stuff as a template if it is useful)?

gavanderhoorn, 2016-12-30:
@NikolausDemmel wrote:
> What has changed is that I have since added support to add custom node packages and have them automatically updated when mounted during `docker run` (useful for development).

That would be interesting, as I was just trying to figure out a nice way to do this myself.

If you have a chance to update the gist or push this to a repository I'd be grateful.

> Maybe @chaoflow wants to maintain a base image at some point (feel free to use our stuff as a template if it is useful)?

@chaoflow: having a Docker image available would definitely make setting up Marv even easier. I'd take up @NikolausDemmel's offer :)

NikolausDemmel, 2017-01-10:
I didn't get around to cleaning this up properly and putting it in a repo. But here is a dump of our current setup for your reference: https://gist.github.com/NikolausDemmel/9248dcde78a4d725cda8cdee32ed67fe

Feel free to take the bits that are useful to you.

gavanderhoorn, 2017-01-16:
Thanks. I'll take a look.