ternaris/marv-robotics #13: No module named marv_cli

dejanpan, 2017-01-31:
On our instance of MARV we all of sudden started getting the following error:
```
$ source venv/bin/activate
(venv)ubuntu@ip-172-20-1-243:~$ cd site/
(venv)ubuntu@ip-172-20-1-243:~/site$ marv --help
Traceback (most recent call last):
  File "/home/ubuntu/venv/bin/marv", line 7, in <module>
    from marv_cli import cli
ImportError: No module named marv_cli
(venv)ubuntu@ip-172-20-1-243:~/site$ 
```

Without any modifications to the host machine. Any suggestions on how to debug/what to do?

dejanpan, 2017-01-31:
I installed venv2 and reinstalled marv and it all works again.

diff --brief -r venv venv2/ points at the difference of 5000 files. 
Any particulars that we should look into?

chaoflow, 2017-02-01:
- check the bin directory, especially `bin/marv`, `bin/python` and `bin/python2.7`
- does marv and marv_cli exist in both `lib/python2.7/site-packages`?

chaoflow, 2017-11-17:
Closing this under the assumption of being caused by a broken virtualenv.

closed 2017-11-17 by chaoflow