ternaris/marv-robotics #15: Requests take up to 20 seconds at 100% load

Sekuraz, 2017-02-13:
After some requests to the overview site this site takes up to 20 seconds to generate on the server producing 100% load. The htop load indicator is red which shows that all cycles are spent in system calls.

Setting the thread count to 1 in uwsgi.conf circumvents this issue. This might be related to [this](https://github.com/unbit/uwsgi/issues/779), at least the "solution" from there works.

We were not able to reproduce this behaviour on another machine.

`$ venv/bin/uwsgi --version
2.0.14

$ cat uwsgi.conf 
[uwsgi]
http = 0.0.0.0:8000
wsgi-file = /opt/bagbunker/site/wsgi.py
; Currently only one process supported
processes = 1
threads = 1
`
We are using the default marv configuration

Sekuraz, 2017-02-14:
It is not infrequent at all, for us it happened every time

Sekuraz, 2017-02-20:
Setting the thread count to 1 has consequences. Marv is only able to deliver as many files simultaneously as there are threads. So fixing this is really important.

chaoflow, 2017-11-17:
@Sekuraz This code path is gone and the new one is designed with issues like this in mind.

closed 2017-11-17 by chaoflow