ternaris/marv-robotics #16: Feature request: Process splitted bagfiles as one

miquelmassot, 2017-06-22:
Is common in ROS to use rosbag record --split command, to store large bagfiles in smaller and manageable sized files. It would be nice to detect these bagfiles (same filename, followed by an underscore and its ordering number) and show topics, information, trajectory, as if it was one.

NikolausDemmel, 2017-06-22:
MARV should already be doing that.

miquelmassot, 2017-06-23:
Well, at least on my side its not happening. I have my bagfiles in folders with:
bagfiles/Day1/YYMMDD_HHMM_0
bagfiles/Day1/YYMMDD_HHMM_1  # same date as previous
bagfiles/Day1/YYMMDD_HHMM_2  # same date as previous 
bagfiles/Day1/YYMMDD_HHMM_0  # new date
bagfiles/Day1/YYMMDD_HHMM_1  # same date as previous 
bagfiles/Day2/...
and so on. 

I see these files listed one by one in the webserver. Is there anything I can report or do?

[image]

NikolausDemmel, 2017-06-23:
As a naming convention it expects the format that `rosbag record` outputs by default, i.e.

```
foo_YYYY-MM-DD-HH-MM-SS_0
foo_YYYY-MM-DD-HH-MM-SS_1
foo_YYYY-MM-DD-HH-MM-SS_2
...
```

See the regex in the code: https://github.com/ternaris/marv-robotics/blob/master/marv_robotics/bagset.py#L38-L39

chaoflow, 2017-11-17:
@miquelmassot The problem here is indeed the naming convention. In the 17.11 this regex has [moved](https://github.com/ternaris/marv-robotics/blob/34c18d90ce150a96a3d82219f47c7f0630634039/marv_robotics/bag.py#L72) and it is straight forward to duplicate / adjust the scanner.

See also the documentation on scanners:
- https://ternaris.com/marv-robotics/docs/config.html#cfg-c-scanner
- https://ternaris.com/marv-robotics/docs/api/marv.html#module-marv.scanner
- https://ternaris.com/marv-robotics/docs/scan.html#ros-bag-file-sets

closed 2018-02-09 by chaoflow