ternaris/marv-robotics #18: Missing Documentation

luator, 2017-07-18:
I am currently looking into MARV Robotics and while it looks very promising, I am lacking some basic documentation on how to implement nodes for custom visualizations (e.g. for showing odometry or laser scan data or getting the plots on the overview page as it is all included in the [demo](https://ternaris.com/marv-robotics/demo)).

Does this documentation exist somewhere? If not, this is something really missing for the project.

NikolausDemmel, 2017-07-18:
@luator, AFAIK, no it does not exist for the current beta version except for the examples in form of the available nodes. And you are absolutely right of course, that this is missing.

In recent months there have been significant developments that we were able to take advantage of as EE customers, but those are not yet reflected in the CE beta yet. Those developments also include some amount of tutorials and API documentation. I believe that an updated CE release is planned some time in the not too distant future, but I have no idea about a concrete plans or timeline, @chaoflow?

chaoflow, 2017-07-18:
@luator The demo is running an Enterprise Edition and documentation for EE is available at https://marvhub.com/docs/ There will be documentation for CE as part of the upcoming release, which will not contain a plotting library. However, it will be possible to load custom widgets (see https://github.com/ternaris/marv/issues/13)

luator, 2017-07-19:
Thanks a lot. For now I will play around a bit with the CE but I guess the EE documentation can help there as well to get a general understanding.

chaoflow, 2017-11-17:
Improved documentation as part of the 17.11 release now [online](https://ternaris.com/marv-robotics/docs/). The documentation is common to CE and EE and it is indicated if features are only available in EE.

closed 2017-11-17 by chaoflow