ternaris/marv-robotics #19: Error: JavaScript ES2015 required?

ugocupcic, 2017-07-18:
Hi,

I'm getting the following error in my browser when trying to connect to MARV - which I installed in Docker using the [following config](https://github.com/shadow-robot/sr-build-tools/blob/marv_robotics/docker/marv_robotics/Dockerfile).

```
MARV requires JavaScript ES2015, which your current browser does not support.
```

I tried accessing it from Google Chrome, Firefox and Safari.

Any idea?

Cheers

NikolausDemmel, 2017-07-18:
Which versions of those browsers and which version of marv specifically? I know this error from too old browsers. Also, I believe Ternaris is developing the frontend with a fairly recent JS standard and then down-compiling the JS to a lower standard during release, so depending on the exact release the requirements might vary.

ugocupcic, 2017-07-18:
Hi @NikolausDemmel, I'm using the latest Google Chrome as far as I know `Version 59.0.3071.115 (Official Build) (64-bit)`

Regarding the version of marv, it's the latest installed from pip on ros-kinetic:
```
Name: marv-robotics
Version: 2.0.0b2
```

NikolausDemmel, 2017-07-18:
Interesting... I can reproduce the issue with your Dockerfile, but when I start an interactive Kinetic container

```
docker run -it -p 8000:8000 --name test ros:kinetic bash
```

and follow the install instructions from the Readme to install 2.0.0b2 from pip

```
apt update
apt-get install curl                   ros-kinetic-ros-base                   python-cv-bridge                   python2.7-dev                   python-opencv                   python-virtualenv                   libjpeg-dev                   libz-dev
rosdep update
source /opt/ros/kinetic/setup.bash
mkdir marv
cd marv/
virtualenv -p python2.7 --system-site-packages venv
source venv/bin/activate
pip install -U pip setuptools
pip install -U marv-robotics
pip list | grep marv # check marv version installed
marv init ./site
cd site/
pip install uwsgi
uwsgi --ini uwsgi.conf
```

I don't get the error, so maybe that helps to find what you are doing differently that might cause this... Have you tried the setup with venv?

ugocupcic, 2017-07-18:
Indeed running it like this seems to fix the issue. Not sure what went wrong without the virtual env.

Thanks!

closed 2017-07-18 by ugocupcic