ternaris/marv-robotics #20: WARNING marv._site Unused config sections ['marv']

huangjiangliang, 2017-10-16:
$ marv fileset scan
2017-10-16 13:56:33,936 INFO rospy.topics topicmanager initialized
2017-10-16 13:56:34,154 INFO marv._node Registered nodes in order: ['fileset', u'md5', u'bagset_name', u'bagmeta', u'messages', u'camera_frames', u'diag_count', u'fulltext', u'gnss_plots', u'navsatfix', u'geo_json_trajectory']
2017-10-16 13:56:34,156 WARNING marv._site Unused config sections ['marv']
2017-10-16 13:56:34,162 INFO rospy.core signal_shutdown [atexit]

chaoflow, 2017-11-17:
@huangjiangliang Thank you for reporting this. This is solved in the 17.11 release of MARV Robotics just pushed to the repos.

closed 2017-11-17 by chaoflow

chaoflow, 2017-11-17:
Link to [config file documenation](https://ternaris.com/marv-robotics/docs/config.html)