ternaris/marv-robotics #21: AssertionError: assert topic['msg_type'] == _topic['msg_type'] in marv_robotics/bag.py line 159

Earthwings, 2017-12-19:
I get the assertion quoted below during "marv run". The topics in question are tf2_msgs/TFMessage vs tf/tfMessage.

```
2017-12-19 08:25:31,592 ERRO marv.cli Exception occured for dataset eb76msjagwhnqhmn5tql4c2rm4:
Traceback (most recent call last):
  File "/home/marv/marv-17-11/venv/local/lib/python2.7/site-packages/marv/cli.py", line 390, in marvcli_run
    cachesize=cachesize)
  File "/home/marv/marv-17-11/venv/local/lib/python2.7/site-packages/marv/site.py", line 355, in run
    deps=deps, cachesize=cachesize)
  File "/home/marv/marv-17-11/venv/local/lib/python2.7/site-packages/marv_node/run.py", line 75, in run_nodes
    done, send_queue_empty = process_task(current, task)
  File "/home/marv/marv-17-11/venv/local/lib/python2.7/site-packages/marv_node/run.py", line 421, in process_task
    return loop()
  File "/home/marv/marv-17-11/venv/local/lib/python2.7/site-packages/marv_node/run.py", line 254, in loop
    promise = current.send(send)
  File "/home/marv/marv-17-11/venv/local/lib/python2.7/site-packages/marv_node/driver.py", line 102, in _run
    request = gen.send(send)
  File "/home/marv/marv-17-11/venv/local/lib/python2.7/site-packages/marv_node/node.py", line 247, in invoke
    send = yield gen.send(send)
  File "/home/marv/marv-17-11/venv/local/lib/python2.7/site-packages/marv_robotics/bag.py", line 159, in bagmeta
    assert topic['msg_type'] == _topic['msg_type']
AssertionError
```

NikolausDemmel, 2018-01-02:
Looks like there are mutliple message types on the same topic, which I guess is not currently supported by marv.

Is it normal and expected that multiple message types should be present on the tf topic? Do the other ROS tools handle this correctly?

chaoflow, 2018-01-11:
@Earthwings Is your use case restricted to compatible message types like ``tf2_msgs/TFMessage`` and ``tf/tfMessage`` on one topic or also mixing of message types, like images and gps on one topic? Currently, we only support one message type for all connections to a topic and are investigating whether and how we can accommodate for different message types.

Earthwings, 2018-01-12:
Right, it is just about compatible messages.

I understand that mixing message types is not supported. Still I would like marv not to crash on such data, but instead ignore the topic and report a warning.

closed 2018-02-09 by chaoflow