ternaris/marv-robotics #22: fulltext.py fails on some messages

Earthwings, 2017-12-19:
During "marv run" I get the following exception:

```
2017-12-19 12:14:08,824 ERRO marv.cli Exception occured for dataset kqgnkij6usqtghmnmhylsngcym:
Traceback (most recent call last):
  File "/home/marv/marv-17-11/venv/local/lib/python2.7/site-packages/marv/cli.py", line 390, in marvcli_run
    cachesize=cachesize)
  File "/home/marv/marv-17-11/venv/local/lib/python2.7/site-packages/marv/site.py", line 355, in run
    deps=deps, cachesize=cachesize)
  File "/home/marv/marv-17-11/venv/local/lib/python2.7/site-packages/marv_node/run.py", line 75, in run_nodes
    done, send_queue_empty = process_task(current, task)
  File "/home/marv/marv-17-11/venv/local/lib/python2.7/site-packages/marv_node/run.py", line 329, in process_task
    return loop()
  File "/home/marv/marv-17-11/venv/local/lib/python2.7/site-packages/marv_node/run.py", line 254, in loop
    promise = current.send(send)
  File "/home/marv/marv-17-11/venv/local/lib/python2.7/site-packages/marv_node/driver.py", line 102, in _run
    request = gen.send(send)
  File "/home/marv/marv-17-11/venv/local/lib/python2.7/site-packages/marv_node/node.py", line 247, in invoke
    send = yield gen.send(send)
  File "/home/marv/marv-17-11/venv/local/lib/python2.7/site-packages/marv_robotics/fulltext.py", line 61, in fulltext
    words = {x for msg in msgs for x in msg.words if msg is not None}
  File "/home/marv/marv-17-11/venv/local/lib/python2.7/site-packages/marv_robotics/fulltext.py", line 61, in <setcomp>
    words = {x for msg in msgs for x in msg.words if msg is not None}
AttributeError: 'NoneType' object has no attribute 'words'
```

Changing the offending line 61 in fulltext.py to
`words = {x for msg in msgs if msg is not None for x in msg.words}`
seems to work around it.

chaoflow, 2018-01-08:
@Earthwings Thank you for your report! The [current line](https://github.com/ternaris/marv-robotics/blob/34c18d90ce150a96a3d82219f47c7f0630634039/marv_robotics/fulltext.py#L61) is ``words = {x for msg in msgs for x in msg.words}``

- Is it correct that the traceback is from a version where you iterated on the workaround, not current master?
- Could you provide a minimal example bag that triggers this bug?

chaoflow, 2018-01-31:
@Earthwings We are approaching a release. Given an example bag triggering this bug, we'd include a fix. So far I fail to understand what causes the issue you are seeing.

Earthwings, 2018-03-19:
After changing the internal format I don't have the offending bags around anymore. It might have been an empty string in a std_msgs/String topic. Still I'd suggest to close my report for now.

closed 2018-03-19 by Earthwings