ternaris/marv-robotics #23: HTTP Basic Auth not working anymore in 17.11

Earthwings, 2018-01-22:
I have a working Marv Robotics 2.0.0. Access is restricted to known user via HTTP basic auth configured in uwsgi.conf:
```
[uwsgi]
route = .* basicauth:My Realm,/home/marv/site/.htpasswd
```
This works fine, the usual browser http basic auth popup comes up and in the Marv web frontend users can choose to login or not.

Now I'm setting up Marv Robotics 17.11 in parallel to upgrade the existing installation and I'd like to do the same for authentication. The following happens:

- when the basicauth route entry is present in the uwsgi section, the basic auth popup works as usual, but the Marv web frontend asks users to login for anything. The login however does not work, the password field is marked with a red border (error indicator) despite using correct credentials
- when the basicauth route entry is not present, no basic auth popup is coming up as expected. Marv then does not ask for a login on its own. The usual Marv login however now works using the same credentials that fail to work when the basicauth entry is present

Is it intentional to bring up the Marv login popup when HTTP basic auth is activated? Any hints how to debug what's going on? This issue is a showstopper for upgrading for me.

chaoflow, 2018-01-31:
@Earthwings we switched to oauth in marv which uses the same request header key as basic auth. What are your use cases for basic auth?

chaoflow, 2018-01-31:
relates https://github.com/ternaris/marv/issues/15

chaoflow, 2018-01-31:
See https://github.com/ternaris/marv/issues/37 for an alternative to basic auth.

closed 2018-01-31 by chaoflow

Earthwings, 2018-02-02:
The use case is preventing other users from accessing potentially sensitive data.

I'm using the workaround from #37 now which seems to work fine. A config option for that would be highly appreciated. Thanks!