ternaris/marv-robotics #24: 17.11: Bag file downloads stall after 1.0 GB, never finish

Earthwings, 2018-01-22:
I have a working Marv Robotics release 2.0. Now I'm testing a parallel installation of Marv 17.11 which is supposed to replace the old one. When users download bag files, this works fine for bag files smaller than 1.0 GB. Larger files however fail to download.

This seems to be independent of the client system and the type of the browser, i.e. a server issue. wget for example starts fine, e.g.
```
...
Length: 2269539342 (2.1G) [application/octet-stream]
...
2018-01-22 11:18:50 (10.9 MB/s) - Connection closed at byte 1082688881. Retrying.
...

```
This does never finish.

chaoflow, 2018-01-31:
@Earthwings We can't reproduce this. Is there anything else involved in your setup than marv and uwsgi? Are you able to download from marvhub.com, e.g.:

```
curl -OJ https://marvhub.com/marv/api/dataset/4cxebk3yrvoxryd7rwvyqx2nfu/0
```

NikolausDemmel, 2018-01-31:
Actually, this hangs and the times out for me around 1GB as well...

```
$ curl -OJ https://marvhub.com/marv/api/dataset/4cxebk3yrvoxryd7rwvyqx2nfu/0
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
 67 1547M   67 1038M    0     0  13.4M      0  0:01:55  0:01:17  0:00:38     0
curl: Saved to filename '2011-04-06-06-56-58.bag'
curl: (18) transfer closed with 533330833 bytes remaining to read
```

(This is from TUM network. No proxy.)

dseifert, 2018-01-31:
Same here:
```
> curl -OJ https://marvhub.com/marv/api/dataset/4cxebk3yrvoxryd7rwvyqx2nfu/0
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
 67 1547M   67 1039M    0     0  3906k      0  0:06:45  0:04:32  0:02:13     0
curl: Saved to filename '2011-04-06-06-56-58.bag'
curl: (18) transfer closed with 533015777 bytes remaining to read
```

chaoflow, 2018-02-01:
Thank you for your feedback. The error does not occur on marvhub.com when downloading to the same machine, but does occur when downloading to another one. In both cases, it's nginx, uwsgi, and marv.

@Earthwings Are you using nginx as well?

chaoflow, 2018-02-01:
The quick solution is to increase [uwsgi_max_temp_file_size](http://nginx.org/en/docs/http/ngx_http_uwsgi_module.html#uwsgi_max_temp_file_size) to have room for your biggest bag file. The proper solution involves X-Sendfile/X-Accel-Redirect which we'll follow up on.

Earthwings, 2018-02-01:
@chaoflow Right, I'm using nginx for https support.

For the marvhub.com file I get the same timeout / early close as reported above.

Earthwings, 2018-02-02:
@chaoflow I'm using the uwsgi_max_temp_file_size workaround now, thanks for sharing. Works fine.

chaoflow, 2018-03-05:
@Earthwings Another solution is to disable buffering completely: http://nginx.org/en/docs/http/ngx_http_uwsgi_module.html#uwsgi_buffering

chaoflow, 2018-03-06:
disabled buffering is documented for upcoming 3.2 release

closed 2018-03-10 by chaoflow