ternaris/marv-robotics #25: Error when a bag misses a topic / tutorial does not work

sebastianscherer, 2018-01-29:
Follow the tutorial [Tutorial: Write your own nodes](https://ternaris.com/marv-robotics/docs/tutorial/write-your-own.html) until 'Display gallery of images'

Download a bagfile that does not include the topic '/wide_stereo/left/image_rect_throttle', e.g. 'https://marvhub.com/marv/api/dataset/kghkhkgnm5jgjuzsaeu6xfppei/0' to the scanroot, run 'marv scan' and 'marv run --collection=bags'. The last command fails because the new bag does not contain the topic.

This crashes the whole run, which should not happen.

Error message:
```
$ marv run --collection=bags
2018-01-29 13:59:29,788 INFO rospy.topics topicmanager initialized
2018-01-29 13:59:30,059 INFO marv.run 2pq6bff4ag.image_section.io4thnkdxx.default (image_section) started
2018-01-29 13:59:30,060 INFO marv.run 2pq6bff4ag.image.og54how3rb.default (image) started
2018-01-29 13:59:30,062 ERRO marv.cli Exception occured for dataset 2pq6bff4agxwyk576ct5uqwnde:
Traceback (most recent call last):
  File "/home/user/.virtualenvs/test/local/lib/python2.7/site-packages/marv/cli.py", line 390, in marvcli_run
    cachesize=cachesize)
  File "/home/user/.virtualenvs/test/local/lib/python2.7/site-packages/marv/site.py", line 355, in run
    deps=deps, cachesize=cachesize)
  File "/home/user/.virtualenvs/test/local/lib/python2.7/site-packages/marv_node/run.py", line 76, in run_nodes
    after()
  File "/home/user/.virtualenvs/test/local/lib/python2.7/site-packages/marv_node/run.py", line 426, in after_loop
    assert not any(driver for lst in waiting.values()
AssertionError
2018-01-29 13:59:30,063 ERRO marv.cli Error occured for dataset 2pq6bff4agxwyk576ct5uqwnde: 
```

beetleskin, 2018-01-29:
see https://github.com/ternaris/marv/issues/29

chaoflow, 2018-02-03:
@sebastianscherer We failed to test the tutorial in real-world scenarios. Usually we use ``foreach`` which works around that problem. It is yet undocumented, as it will undergo a change in one of the next releases. ``foreach`` leads to the node being instantiated for each stream of an input. In case of the raw_messages node as an input, that means for every topic matching the filter passed to ``marv.select``.

In case of the image node of the tutorial, try this:

```python
@marv.node(File)
@marv.input('cam', foreach=marv.select(raw_messages, TOPIC))
def image(cam)
```

and the consuming section needs then another layer of pulling:

```python
    # fetch first image stream, corresponding to one topic
    img_stream = yield marv.pull(image)
    if img_stream is None:
        return
    # fetch first image from first image stream
    img = yield marv.pull(img_stream)
    if img is None:
        return
```

See also:
- https://github.com/ternaris/marv-robotics/blob/master/marv_robotics/cam.py#L38
- https://github.com/ternaris/marv-robotics/blob/master/marv_robotics/detail.py#L228

Note: The ``node[filter]`` syntax is old for ``marv.select(node, filter)``

beetleskin, 2018-03-06:
So if I got this right: for the sake of getting rid of that error and actively support optional topics, I have to introduce an extra layer using the (unsupported) `folreach` param, which unpacks the individual messages, puts them into an array, pushes them with `yield marv.push({'values': my_decodec_data_array})` as stream. This stream is then consumed by the actual node which decodes this array again, pulls twice (but only if data is there) and then I can work with the data?

From https://github.com/ternaris/marv/issues/29 you suggested to continue here, so I will stick to that example. My section-node was originally:

```python
@marv.node(Section)
@marv.input('title', default='User Annotations')
@marv.input('stream', marv.select(raw_messages, "/user_annotation"))
def annotations_section(title, stream):
    pytype = get_message_type(stream)
    rosmsg = pytype()
    annotations = []

    while True:
        msg = yield marv.pull(stream)
        if msg is None:
            break
        rosmsg.deserialize(msg.data)
        annotations += [copy.deepcopy(rosmsg)]
    
  # .. create a fancy widget-table with annotations list
``` 

This works nicely when the topic is available. If not, the known error is generated and `marv run stops at this point`. If I got this right (from your suggestions here and from the [tutorial](https://ternaris.com/marv-robotics/docs/_modules/marv_robotics/gnss.html)), you suggest to do the following:

```python
@marv.node()
@marv.input('stream', foreach=marv.select(raw_messages, '/user_annotation:*'))
def _annotations_collector(stream):
    yield marv.set_header(title=stream.topic)
    annotations = []
    pytype = get_message_type(stream)
    rosmsg = pytype()

    while True:
        proto_msg = yield marv.pull(stream)
        if proto_msg is None:
            break
        rosmsg.deserialize(proto_msg.data)
        annotations.append((copy.deepcopy(rosmsg)))

    yield marv.push({'values': annotations})

@marv.node(Section)
@marv.input('title', default='User Annotations')
@marv.input('streams', default=_annotations_collector)
def annotations_section(title, stream):
    annotations = []
    while True:
        stream = yield marv.pull(streams)
        if stream is None:
            break
        yet_another_stream = yield marv.pull(stream)
        if yet_another_stream is None:
            break
        annotations += yet_another_stream['values']
    
  # .. create a fancy widget-table with annotations list
```

---

I can confirm that this works. But:
* We have the situation that our bags are "corrupt" in lets say 30% of all cases, which is okay and won't change (btw. this was always the case when using bags in projects/universities/companies in my past). We want to scrape, collect and display the data that is there, rather than abandon bags which are incomplete. That in mind, it would be a huge overhead to guard every _marv-node_ with this _foreach_-workaround.
* The stream-ception confuses me a lot: I never know if I got a Handle, a Pull, a node or an array of the aforementioned. I don't know where to pull and check against None.
* Unwrapping a topic in this collector-node seems to contradict the whole capnp-serialization resp. the inter-node-communication idea.

Isn't there a simpler way? In the older Marv-version, missing input topics just didn't trigger the Node consuming it. Why did you change that behavior to generate an error? Afaik, with this you force the Marv-User to have bags that consist of the same topics .. or live with the error.

chaoflow, 2018-07-22:
@sebastianscherer [optional inputs](https://ternaris.com/marv-robotics/docs/patterns.html#optional-inputs) are now supported and the nodes from the tutorial should now also work with bag files lacking the selected topics.

chaoflow, 2018-07-22:
@beetleskin Apart from a minor adjustment your original section node should work, see [optional inputs](https://ternaris.com/marv-robotics/docs/patterns.html#optional-inputs).

Regarding 2: Thank you for the feedback on that. You got in contact with (undocumented) groups due to the (undocumented) foreach workaround. Both should not be needed anymore for optional inputs. Pulling individual topics should now be straight forward. A node always receives handles for streams.  In case of an individual topic, the stream will be a stream of messages, i.e. pulling will give you one message after another. (If you use wildcards in select as in the foreach workaround, you will receive a group stream, i.e. a stream of streams. Pulling such a stream gives you handles to the streams contained in the group.)

Regarding 3 and 4: From your code snippet I take, that you want to display a list of annotations. To achieve this ros messages need to be deserialised and the values of all/selected annotation messages need to be gathered into a list. The list of messages is reduced to one message with a list of (selected) annotation values. In order not to keep all ros messages in memory we [reduce separately](https://ternaris.com/marv-robotics/docs/patterns.html#reduce-separately) for each stream/topic.

If this does not answer your points, please open new dedicated issues accordingly.

closed 2018-07-22 by chaoflow