ternaris/marv-robotics #28: Corrupted GPS messages crash marv

sebastianscherer, 2018-02-12:
If you perform `marv run` on a bag with corrupted gps messages (the topic has no valid message), marv crashes with this error:

```
$ marv run --collection=bags
2018-02-12 14:43:51,547 INFO rospy.topics topicmanager initialized
2018-02-12 14:43:51,815 INFO marv.run io6slfamtm.gnss_section.kapv24qfhd.default (gnss_section) started
2018-02-12 14:43:51,816 INFO marv.run io6slfamtm.gnss_plots.phjlglnopu.default (gnss_plots) started
2018-02-12 14:44:24,630 WARN marv.node.io6slfamtm.positions.ufturrk7wi.0 skipped 2268 erroneous messages
2018-02-12 14:44:24,771 ERRO marv.cli Exception occured for dataset io6slfamtmbfj6gemdaku57xnu:
Traceback (most recent call last):
  File "/home/username/.virtualenvs/marv/local/lib/python2.7/site-packages/marv/cli.py", line 390, in marvcli_run
    cachesize=cachesize)
  File "/home/username/.virtualenvs/marv/local/lib/python2.7/site-packages/marv/site.py", line 355, in run
    deps=deps, cachesize=cachesize)
  File "/home/username/.virtualenvs/marv/local/lib/python2.7/site-packages/marv_node/run.py", line 75, in run_nodes
    done, send_queue_empty = process_task(current, task)
  File "/home/username/.virtualenvs/marv/local/lib/python2.7/site-packages/marv_node/run.py", line 375, in process_task
    return loop()
  File "/home/username/.virtualenvs/marv/local/lib/python2.7/site-packages/marv_node/run.py", line 254, in loop
    promise = current.send(send)
  File "/home/username/.virtualenvs/marv/local/lib/python2.7/site-packages/marv_node/driver.py", line 102, in _run
    request = gen.send(send)
  File "/home/username/.virtualenvs/marv/local/lib/python2.7/site-packages/marv_node/node.py", line 247, in invoke
    send = yield gen.send(send)
  File "/home/username/.virtualenvs/marv/local/lib/python2.7/site-packages/marv_robotics/gnss.py", line 219, in gnss_plots
    gps = gps[np.isfinite(gps[:, 1])]
IndexError: too many indices for array
2018-02-12 14:44:24,771 ERRO marv.cli Error occured for dataset io6slfamtmbfj6gemdaku57xnu: too many indices for array
```

I believe marv should catch this and either display an error message on the website or simply skip gnss for this bag file instead.

sebastianscherer, 2018-02-12:
This happens for messages of type `sensor_msgs/NavSatFix` that contain nans:

```
header:
  seq: 3538
  stamp:
    secs: 1516381073
    nsecs: 150760810
  frame_id: "navsat_link"
status:
  status: -1
  service: 1
latitude: nan
longitude: nan
altitude: nan
position_covariance: [nan, 0.0, 0.0, 0.0, nan, 0.0, 0.0, 0.0, nan]
position_covariance_type: 1
```

chaoflow, 2018-03-07:
@sebastianscherer we are [already checking](https://github.com/ternaris/marv-robotics/blob/master/marv_robotics/gnss.py#L74) a couple of fields for ``nan`` . My assumption is that a message with non-nan lon/lat/alt but nan ``rosmsg.position_covariance[0]`` causes the issue.

Could you provide an example bag triggering the bug or verify whether the following fixes the issue:
```
        if not hasattr(rosmsg, 'status') or \
           np.isnan(rosmsg.longitude) or \
           np.isnan(rosmsg.latitude) or \
           np.isnan(rosmsg.altitude) or \
           np.isnan(rosmsg.position_covariance[0]):
            erroneous += 1
            continue
```
https://github.com/ternaris/marv-robotics/blob/master/marv_robotics/gnss.py#L74

chaoflow, 2018-03-07:
@sebastianscherer reread your initial post. The fix for no valid gps messages should be:
```
-    yield marv.push({'values': positions})
+    if positions:
```
https://github.com/ternaris/marv-robotics/blob/master/marv_robotics/gnss.py#L101

closed 2018-03-10 by chaoflow