ternaris/marv-robotics #29: cam.py crashing when trying to process 16 bit images

sebastianscherer, 2018-03-19:
Hi, 

cam.py currently crashes for us when being fed 16 bit (e.g. depth) images. Example:

```
Traceback (most recent call last):
  File "/home/user/.virtualenvs/marv/local/lib/python2.7/site-packages/marv/cli.py", line 418, in marvcli_run
    excluded_nodes, cachesize=cachesize)
  File "/home/user/.virtualenvs/marv/local/lib/python2.7/site-packages/marv/site.py", line 361, in run
    deps=deps, cachesize=cachesize)
  File "/home/user/.virtualenvs/marv/local/lib/python2.7/site-packages/marv_node/run.py", line 76, in run_nodes
    done, send_queue_empty = process_task(current, task)
  File "/home/user/.virtualenvs/marv/local/lib/python2.7/site-packages/marv_node/run.py", line 330, in process_task
    return loop()
  File "/home/user/.virtualenvs/marv/local/lib/python2.7/site-packages/marv_node/run.py", line 255, in loop
    promise = current.send(send)
  File "/home/user/.virtualenvs/marv/local/lib/python2.7/site-packages/marv_node/driver.py", line 102, in _run
    request = gen.send(send)
  File "/home/user/.virtualenvs/marv/local/lib/python2.7/site-packages/marv_node/node.py", line 256, in invoke
    send = yield gen.send(send)
  File "/home/user/.virtualenvs/marv/local/lib/python2.7/site-packages/marv_robotics/cam.py", line 131, in images
    img = imgmsg_to_cv2(rosmsg, "rgb8")
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/cv_bridge/core.py", line 184, in imgmsg_to_cv2
    raise CvBridgeError(e)
CvBridgeError: [16UC1] is not a color format. but [rgb8] is. The conversion does not make sense
```

This might actually a bug in cv_bridge, but [here is a possible workaround](https://gist.github.com/awesomebytes/30bf7eae3a90754f82502accd02cbb12).

Earthwings, 2018-04-13:
Same here for 16SC1 (16 bit images with Bayer encoding, 12 bit used)

chaoflow, 2018-05-08:
@sebastianscherer We added the linked workaround to handle 16UC1 images. Does that work for you?

chaoflow, 2018-05-08:
@Earthwings What conversion would you expect for 16SC1? Could you provide a small example bag?