ternaris/marv-robotics #30: Upload .txt and yaml files

new-in-ros, 2018-04-23:
Hello,
I'm new in ROS and I have to built a database for ROS-bagfiles. Is it possible to upload .txt and yaml files in the MARV database and how can I do that.
Best regards
new-in-ros

new-in-ros, 2018-04-24:
@chaoflow  thanks for your replay.
We want to have a database with different bagfiles for simulations. There are fore example calibration files (.yaml) of cameras with should also be stored in the data base. 
I want to copy this files in the scanroot directory and load it up into the database.

new-in-ros, 2018-04-26:
@chaoflow thanks for your help. I found a solution witch is ok for me. I made a new collection with an own scanner and it works.

new-in-ros, 2018-07-04:
Hello, 
now I am able to upload all data types. I added some code in `bag.py` and all files are listed in the same collection (bags). I can download the bag-files from the database (in the column `Summary`) after running `marv run --collection=bags`. Unfortunately for the other file types this column is empty after running `marv run --collection=bags`.
Where do I have to change the code that I can download all data types? 
Best regards
new-in-ros

chaoflow, 2018-07-06:
@new-in-ros ``marv run --collection=bags`` runs all nodes configured for the collection named ``bags``. Use the name for the other collection to run nodes for that one or ``marv run --collection=*`` to run for all configured collections. If that does not answer your question, please [provide a minimal working example](https://github.com/ternaris/marv-robotics#reporting-issues--minimal-working-example) so we can reproduce your setup.

new-in-ros, 2018-07-06:
@chaoflow I made a fork an added my example files (.bag, .yaml, .png), the configuration and the `bag.py` file. In the code I just changed the `bag.py` file (line 153-169). The added part allows to upload all data types in the collection `bags`. After running `marv run --collection=bags` I just can download the bag files.
Link:
https://github.com/new-in-ros/marv-robotics

chaoflow, 2018-07-06:
@new-in-ros For the changed ``bag.py`` to be used it needs to be in the same location as the original ``bag.py``

new-in-ros, 2018-07-06:
@chaoflow  Thanks for your fast reply. Now the `bag.py` file is at the right directory. I also changed the configuration files.
https://github.com/new-in-ros/marv-robotics

new-in-ros, 2018-07-09:
@chaoflow thanks for your help.
I wanted to label the different files with tags. But it would be also enough, if the bag files and the other files are separated in different collections.
What do I have to do for implementing one collection for bag files and one for the other files? How does the configuration file look like? 
With the database it should be possible to download all data types. I need it for my bachelor thesis and the database is just a very small part of this thesis. So it doesn't have to be a perfect solution.

chaoflow, 2018-07-09:
@new-in-ros something like this

```
[marv]
collections = foo bar

[collection foo]
...

[collection bar]
...
```

new-in-ros, 2018-07-09:
@chaoflow Thanks for your great support. 
Now I have a good solution. 
Best regards new-in-ros

closed 2018-07-09 by new-in-ros