ternaris/marv-robotics #43: Cleanup discarded datasets does not erase the file

tigelbri, 2018-11-05:
Hi,

at first I want to say a great thanks for that awesome tool. 
As far as I understand the code the 'marv cleanup --discarded' erases db entries only and not the file itself. I think thats an issue due to the reindexing of the old files at the next scan.

chaoflow, 2018-11-05:
@tigelbri Thank you for your feedback! That behaviour is intentional, marv should not have write access to your log files. The intended workflow is to first delete files manually and then discard them, e.g.

```
rm -rf /path/to/some/folder
marv scan
marv query --missing |xargs marv discard
marv cleanup --discarded
```
Does this answer your question?

tigelbri, 2018-11-07:
Yes, thank you for that explanation!

closed 2018-11-07 by chaoflow