ternaris/marv-robotics #44: MARV HTTP API filtering

ggaessler, 2018-12-05:
Filtering via the HTTP API doesn't seem work as intended. 

The following requests
```
curl -k -G --data-urlencode 'filter={"tags": {"op": "any", "val": ["bar", "foo"]}}' $MARV_API/collection
curl -k -G --data-urlencode 'filter={"files": {"op": "substring_any", "val": "2018"}}' $MARV_API/collection
curl -k -G --data-urlencode 'filter={"tags": {"op": "all", "val": "2018"}}' $MARV_API/collection
```
return exactly the same results, a huge amount of message types, topics, tags, ... but not a single dataset. The results seems to be completely independent of filter type and value.

The first example is taken directly from the documentation:
https://ternaris.com/marv-robotics/docs/httpapi.html

chaoflow, 2018-12-06:
@ggaessler For me the first and second are working, in case of the third you should provide a list like in the first example and I consider it a bug that marv does not complain about that - need to take a closer look however. Are you sure you have matching datasets in your marv instance?

ggaessler, 2018-12-07:
@chaoflow thanks! The second and third one were supposed to match files, the first one not, but delivered the same result. I didn't expect the call to return all the other data apart from `rows`. 
The third request now works perfectly fine after turning it into a list. And the second one now also works fine, I guess I was connected to the wrong MARV instance when testing this request. Sorry, my fault.

closed 2018-12-07 by chaoflow