ternaris/marv-robotics #45: marvcli_run crashes when trying to process a bag file containing std_msgs/String with ü, ä or ö

pahemme, 2018-12-12:
2018-12-12 15:46:45,970 ERRO marv.cli Exception occured for dataset wrcb3sjanmbcqb7vfj7hqt53cy: Traceback (most recent call last):
  File "/opt/marv/local/lib/python2.7/site-packages/marv/cli.py", line 432, in marvcli_run
    excluded_nodes, cachesize=cachesize)
  File "/opt/marv/local/lib/python2.7/site-packages/marv/site.py", line 392, in run
    collection.update_listings([dataset])
  File "/opt/marv/local/lib/python2.7/site-packages/marv/collection.py", line 732, in update_listings
    db.session.execute(stmt, listings)
  File "/opt/marv/local/lib/python2.7/site-packages/sqlalchemy/orm/scoping.py", line 153, in do
    return getattr(self.registry(), name)(*args, **kwargs)
  File "/opt/marv/local/lib/python2.7/site-packages/sqlalchemy/orm/session.py", line 1176, in execute
    bind, close_with_result=True).execute(clause, params or {})
  File "/opt/marv/local/lib/python2.7/site-packages/sqlalchemy/engine/base.py", line 948, in execute
    return meth(self, multiparams, params)
  File "/opt/marv/local/lib/python2.7/site-packages/sqlalchemy/sql/elements.py", line 269, in _execute_on_connection
    return connection._execute_clauseelement(self, multiparams, params)
  File "/opt/marv/local/lib/python2.7/site-packages/sqlalchemy/engine/base.py", line 1060, in _execute_clauseelement
    compiled_sql, distilled_params
  File "/opt/marv/local/lib/python2.7/site-packages/sqlalchemy/engine/base.py", line 1200, in _execute_context
    context)
  File "/opt/marv/local/lib/python2.7/site-packages/sqlalchemy/engine/base.py", line 1413, in _handle_dbapi_exception
    exc_info
  File "/opt/marv/local/lib/python2.7/site-packages/sqlalchemy/util/compat.py", line 203, in raise_from_cause
    reraise(type(exception), exception, tb=exc_tb, cause=cause)
  File "/opt/marv/local/lib/python2.7/site-packages/sqlalchemy/engine/base.py", line 1193, in _execute_context
    context)
  File "/opt/marv/local/lib/python2.7/site-packages/sqlalchemy/engine/default.py", line 508, in do_execute
    cursor.execute(statement, parameters)
ProgrammingError: (sqlite3.ProgrammingError) You must not use 8-bit bytestrings unless you use a text_factory that can interpret 8-bit bytestrings (like text_factory = str). It is highly recommended that you instead just switch your application to Unicode strings. [SQL: u'INSERT OR REPLACE INTO listing_wtd_monitoring (id, "row", name, setid, size, fulltext, added_time, start_time, end_time, duration) VALUES (?, ?, ?, ?, ?, ?, ?, ?, ?, ?)'] [parameters: (227, '{"id":227,"setid":"wrcb3sjanmbcqb7vfj7hqt53cy","tags":["#TAGS#"],"values":[{"id":"wrcb3sjanmbcqb7vfj7hqt53cy","route":"detail","title":"2018-10-16-16-52-44"},20670951488,["#TAGS#"],1544625779915,1539701564570,1539702042204,477634]}', u'2018-10-16-16-52-44', u'wrcb3sjanmbcqb7vfj7hqt53cy', 20670951488, 'and ab wieder verliert immer im go zu verz\xc3\xb6gerung springt \xc3\xbcberlenkt (vermutung system manuelle start bj\xc3\xb6rn) auto pose stop jetzt unibw bis tracking 30 m versuch fahrt runde manuel konfigurieren software', 1544625779915, 1539701564570, 1539702042204, 477634)] (Background on this error at: http://sqlalche.me/e/f405)
2018-12-12 15:46:45,995 ERRO marv.cli Error occured for dataset wrcb3sjanmbcqb7vfj7hqt53cy: (sqlite3.ProgrammingError) You must not use 8-bit bytestrings unless you use a text_factory that can interpret 8-bit bytestrings (like text_factory = str). It is highly recommended that you instead just switch your application to Unicode strings. [SQL: u'INSERT OR REPLACE INTO listing_wtd_monitoring (id, "row", name, setid, size, fulltext, added_time, start_time, end_time, duration) VALUES (?, ?, ?, ?, ?, ?, ?, ?, ?, ?)'] [parameters: (227, '{"id":227,"setid":"wrcb3sjanmbcqb7vfj7hqt53cy","tags":["#TAGS#"],"values":[{"id":"wrcb3sjanmbcqb7vfj7hqt53cy","route":"detail","title":"2018-10-16-16-52-44"},20670951488,["#TAGS#"],1544625779915,1539701564570,1539702042204,477634]}', u'2018-10-16-16-52-44', u'wrcb3sjanmbcqb7vfj7hqt53cy', 20670951488, 'and ab wieder verliert immer im go zu verz\xc3\xb6gerung springt \xc3\xbcberlenkt (vermutung system manuelle start bj\xc3\xb6rn) auto pose stop jetzt unibw bis tracking 30 m versuch fahrt runde manuel konfigurieren software', 1544625779915, 1539701564570, 1539702042204, 477634)] (Background on this error at: http://sqlalche.me/e/f405)

chaoflow, 2018-12-30:
@pahemme Thank you for reporting this. Could you please provide minimal working example in form of a bag file and marv.conf triggering this bug?

chaoflow, 2019-01-07:
@pahemme thx!

chaoflow, 2019-01-07:
relates #42

chaoflow, 2019-02-05:
@pahemme This should be fixed with https://github.com/ternaris/marv-robotics/commit/cefde723e41bde65d3579cb33cf57ca1d1966795 Can you confirm?

closed 2019-02-05 by chaoflow

pahemme, 2019-02-07:
Yes, thank you!