ternaris/marv-robotics #54: Migration to GitLab

chaoflow, 2019-02-09:
Consolidating our tools, we're moving MARV Robotics to [GitLab](https://gitlab.com/). We hope you follow us there. Please let us know if this causes you any inconvenience.

During import issues, pull requests, and comments will be associated to your user accounts on Git**Lab**. For this to work, they need to exist before importing.

There are [two options](https://docs.gitlab.com/ee/user/project/import/github.html#using-the-github-integration):
- You log into gitlab using your github Account OR
- You create a gitlab account with an email address that matches the public email address of your github user

We kindly ask you to do this until February 15th. After that we'll do the import and notify you here.

Users having created issues or pull requests that so far do not have a matching gitlab account:
- @ggaessler
- @schra
- @Sekuraz
- @pahemme
- @tigelbri
- @new-in-ros
- @ddimarco
- @huangjiangliang
- @jpsamper2009
- @luator
- @ugocupcic
- @mikepurvis

ddimarco, 2019-02-11:
My gitlab account is https://gitlab.com/ddimarco

ggaessler, 2019-02-11:
Mine is https://gitlab.com/ggaessler

chaoflow, 2019-02-22:
Thank you to everybody who created an account.

@josephduchesne @gavanderhoorn @tigelbri @beetleskin @dseifert You contributed after the initial round or we missed to notify you initially. Please see https://github.com/ternaris/marv-robotics/issues/54#issue-408479389 if you want your contributions (issue, comment, pull request) to be associated with your git**lab** account.

beetleskin, 2019-03-05:
https://gitlab.com/beetleskin

Why exactly are you moving?

gavanderhoorn, 2019-03-05:
https://gitlab.com/gavanderhoorn.

chaoflow, 2019-03-05:
Migration is complete now: https://gitlab.com/ternaris/marv-robotics

@beetleskin We waited with the import for people to create their accounts and are moving to consolidate our tools. -- see https://github.com/ternaris/marv-robotics/issues/54#issue-408479389

closed 2019-03-05 by chaoflow

ternaris-gitlab-ci, 2019-03-06:
@gavanderhoorn @ddimarco @jpsamper2009 You seem to have created users with the same username but different primary email address. The matching primary email address is required if the account was not created by logging into gitlab using the github account, i.e. the gitlab icon, i.e. oauth.

In case you want to adjust that, we'll re-import after 2019-03-07 12:00 CET.

gavanderhoorn, 2019-03-06:
@ternaris-gitlab-ci: I'm pretty sure the primary email addresses for both accounts are identical.

Can you link to a commit in this repository which you can't map to the gitlab account?

chaoflow, 2019-03-06:
@gavanderhoorn Thanks for the quick reply. Here is a comment were gitlab failed to associate: https://gitlab.com/ternaris/marv-robotics/issues/12#note_147708841

If the email addresses are the same I'll contact gitlab support again.

gavanderhoorn, 2019-03-06:
Ah, so that's a comment, not a commit.

I've just verified, and both my GH and GL accounts have the same address set as primary.

chaoflow, 2019-03-06:
@gavanderhoorn The issue is that the email address for your GH account is not publicly visible, @ddimarco @jpsamper2009 same for you.

gavanderhoorn, 2019-03-06:
> @gavanderhoorn The issue is that the email address for your GH account is not publicly visible, 

that's actually on purpose.

If that is what is required then I'm ok with my contributions not being attributed to me on GL.

chaoflow, 2019-03-06:
@gavanderhoorn another possibility would be to delete your GL account and re-create it by using the GH icon on the GL login page.

@ddimarco @jpsamper2009 If I don't hear from you, I'll wait until tomorrow 2019-03-07 12:00 CET for the final import to GL.